sieren/Homepoint #180: M5Stack Core2 does not show any files

DGrebe, 2021-12-14:
Hi,

this project is great work, I really appreciate your work.
 
I have two devices: an AZ Touch and a M5Stack Core2.

For the AZ Touch everything works fine but for the M5Stack Core2 I can initialize the system and get the HomePoint screen, but the file area is completely empty. I worked with the binaries from the release as well as with own build binaries and get the same results.

I have also uploaded the individual bin files to the following locations, same result.
> esptool.py --port /dev/cu.SLAB_USBtoUART --chip esp32 -b 921600 write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 bootloader.bin
> esptool.py --port /dev/cu.SLAB_USBtoUART --chip esp32 -b 921600 write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x8000 partition-table.bin
> esptool.py --port /dev/cu.SLAB_USBtoUART --chip esp32 -b 921600 write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0xd000 ota_data_initial.bin
> esptool.py --port /dev/cu.SLAB_USBtoUART --chip esp32 -b 921600 write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x10000 homepoint.bin
> esptool.py --port /dev/cu.SLAB_USBtoUART --chip esp32 -b 921600 write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x2b0000 spiffs.bin

When I reload the home page, the M5Stack Core2 restarts, giving the following lines in the serial monitor.
It complains about the file system, which is not mounted. Looks like a problem with a wrong or incorrectly mounted SPIFFS.

----
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060030  A0      : 0x800f0505  A1      : 0x3ffd24a0  
A2      : 0x80008000  A3      : 0x80007ffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffd2470  
A10     : 0x3ffd24d0  A11     : 0x00000000  A12     : 0x0000000b  A13     : 0x3ffd23a0  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x80008000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: baad76808584ee02

Backtrace: 0x400014fa:0x3ffd24a0 0x400f0502:0x3ffd24b0 0x400f6cd1:0x3ffd24d0 0x400f6da1:0x3ffd2500 0x400f6eb5:0x3ffd2540 0x400f7099:0x3ffd2580 0x40100769:0x3ffd25a0 0x401007e5:0x3ffd25d0 0x40100ef6:0x3ffd25f0

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6680
load:0x40078000,len:12636
load:0x40080400,len:3792
entry 0x40080684
[0;32mI (61) boot: Chip Revision: 3[0m
[0;32mI (65) boot_comm: chip revision: 3, min. bootloader chip revision: 0[0m
[0;32mI (39) boot: ESP-IDF v4.0.3-dirty 2nd stage bootloader[0m
[0;32mI (40) boot: compile time 21:33:16[0m
[0;32mI (40) boot: Enabling RNG early entropy source...[0m
[0;32mI (45) boot: SPI Speed      : 80MHz[0m
[0;32mI (49) boot: SPI Mode       : DIO[0m
[0;32mI (53) boot: SPI Flash Size : 4MB[0m
[0;32mI (57) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (75) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (90) boot:  3 ota_0            OTA app          00 10 00010000 00150000[0m
[0;32mI (98) boot:  4 ota_1            OTA app          00 11 00160000 00150000[0m
[0;32mI (105) boot:  5 spiffs           Unknown data     01 82 002b0000 000fa000[0m
[0;32mI (113) boot: End of partition table[0m
[0;32mI (117) boot_comm: chip revision: 3, min. application chip revision: 0[0m
[0;32mI (124) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x4754c (292172) map[0m
[0;32mI (228) esp_image: segment 1: paddr=0x00057574 vaddr=0x3ffb0000 size=0x03784 ( 14212) load[0m
[0;32mI (233) esp_image: segment 2: paddr=0x0005ad00 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (235) esp_image: segment 3: paddr=0x0005b108 vaddr=0x40080400 size=0x04f08 ( 20232) load[0m
[0;32mI (251) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xdbea8 (900776) map[0m
[0;32mI (544) esp_image: segment 5: paddr=0x0013bec8 vaddr=0x40085308 size=0x0fa60 ( 64096) load[0m
[0;32mI (581) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (581) boot: Disabling RNG early entropy source...[0m
[0;32mI (582) cpu_start: Pro cpu up.[0m
[0;32mI (585) cpu_start: Application information:[0m
[0;32mI (590) cpu_start: Project name:     homepoint[0m
[0;32mI (595) cpu_start: App version:      v0.07.2-10-g9948fc4-dirty[0m
[0;32mI (602) cpu_start: Compile time:     Dec 13 2021 21:33:43[0m
[0;32mI (608) cpu_start: ELF file SHA256:  baad76808584ee02...[0m
[0;32mI (614) cpu_start: ESP-IDF:          v4.0.3-dirty[0m
[0;32mI (619) cpu_start: Starting app cpu, entry point is 0x4008135c[0m
[0;32mI (606) cpu_start: App cpu up.[0m
[0;32mI (630) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (637) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (643) heap_init: At 3FFBB150 len 00024EB0 (147 KiB): DRAM[0m
[0;32mI (649) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (655) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (662) heap_init: At 40094D68 len 0000B298 (44 KiB): IRAM[0m
[0;32mI (668) cpu_start: Pro cpu start user code[0m
[0;32mI (688) spi_flash: detected chip: generic[0m
[0;32mI (689) spi_flash: flash io: dio[0m
[0;33mW (689) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.[0m
[0;32mI (699) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[E][esp32-hal-cpu.c:125] removeApbChangeCallback(): not found func=400F34A0 arg=3FFB13A8
[E][vfs_api.cpp:22] open(): File system is not mounted
 Jpeg file not found
[E][vfs_api.cpp:22] open(): File system is not mounted
 Jpeg file not found
[0;31mE (2013) example: Failed to find Font[0m
I (2043) wifi:wifi driver task: 3ffde7fc, prio:23, stack:6656, core=0
[0;32mI (2043) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (2048) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (2069) wifi:wifi firmware version: 52d2120
I (2069) wifi:config NVS flash: enabled
I (2069) wifi:config nano formating: disabled
I (2071) wifi:Init data frame dynamic rx buffer num: 32
I (2076) wifi:Init management frame dynamic rx buffer num: 32
I (2081) wifi:Init management short buffer num: 32
I (2086) wifi:Init dynamic tx buffer num: 32
I (2090) wifi:Init static rx buffer size: 1600
I (2094) wifi:Init static rx buffer num: 10
I (2098) wifi:Init dynamic rx buffer num: 32
[0;32mI (2103) wifi_init: rx ba win: 6[0m
[0;32mI (2107) wifi_init: tcpip mbox: 32[0m
[0;32mI (2111) wifi_init: udp mbox: 6[0m
[0;32mI (2115) wifi_init: tcp mbox: 6[0m
[0;32mI (2119) wifi_init: tcp tx win: 5744[0m
[0;32mI (2123) wifi_init: tcp rx win: 5744[0m
[0;32mI (2127) wifi_init: tcp mss: 1436[0m
[0;32mI (2131) wifi_init: WiFi IRAM OP enabled[0m
[0;32mI (2135) wifi_init: WiFi RX IRAM OP enabled[0m
[0;32mI (2143) phy_init: phy_version 4660,0162888,Dec 23 2020[0m
I (2227) wifi:mode : sta (b8:f0:09:c6:1f:c0)
[0;31mE (2228) event: system event loop not initialized via esp_event_loop_init[0m
[0;32mI (2228) wifi station: wifi_init_sta finished.[0m
[0;32mI (2234) wifi station: connect to ap SSID:<SSID> password:<password>[0m
I (2267) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2268) wifi:state: init -> auth (b0)
I (2278) wifi:state: auth -> assoc (0)
I (2285) wifi:state: assoc -> run (10)
I (2309) wifi:connected with HOMELAN, aid = 4, channel 11, BW20, bssid = 08:96:d7:80:dc:c6
I (2310) wifi:security: WPA2-PSK, phy: bgn, rssi: -54
I (2314) wifi:pm start, type: 1

[0;31mE (2314) event: system event loop not initialized via esp_event_loop_init[0m
I (2471) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[0;31mE (2527) event: system event loop not initialized via esp_event_loop_init[0m
[0;32mI (2527) tcpip_adapter: sta ip: 192.168.123.10, mask: 255.255.255.0, gw: 192.168.123.254[0m
[0;32mI (2532) wifi station: got ip:192.168.123.10[0m

----

What can I do?

Kind Regards

Dirk

Next9999, 2022-10-14:
Hello

i am looking to use a M5 Core 2 .... is it working on your end?

DGrebe, 2022-10-16:
Hi,

I am not sure anymore, how far I came, as I worked on this last winter. I think I got it running on the M5 Core 2, but it might be unstable.
I have tried to extend the navigation to include folders / hierachies but was not successful, then I stopped working on this topic. I don't have Homepoint active at the moment.

I will check, when I have time to look into it again.

Regards
Dirk

closed 2025-12-05 by DGrebe