sieren/Homepoint #181: Upload to ESP32-Wroom

DSRocker, 2021-12-20:
Hi,
I tried to flash (with ESP-Flasher-Windows-x64) the homepoint_aztouchmod_full.bin to my ESP32-WROOM. 

This is the Log:
Using 'COM4' as serial port.
Connecting....
Detecting chip type... ESP32
Connecting......

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 78:E3:6D:09:81:B8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)

Thanks!

MBadberg, 2021-12-29:
Have you tried the espgeneric file?

I've tried today both, and they worked.

DSRocker, 2021-12-30:
It works fine with ESP32 Download Tool. 
Thanks a lot
Another question: is it possible to connect / to wire 3,5" TFT LCD for arduino uno shield (red) ILI9481 [https://www.ebay.de/itm/174949498297?var=474095516573](url)